KTH-RPL/dufomap #2: nothing different with the gt_point.pcd?

sanduan168, 2024-04-23:
Hi,  Kin
I run the code you pushed 4 days ago with configs in assets/.  But the result dufomap_output.pcd seems no different with the gt_cloud.pcd. The dynamic points are still saved.  Is it right?  
Here is the screentshot:
[image]

Kin-Zhang, 2024-04-23:
could you copy the output of this command?

```bash
uname -a
```

sanduan168, 2024-04-23:
Sure.
The output of the uname command is like below:
`Linux airs-host 5.4.0-150-generic #167~18.04.1-Ubuntu SMP Wed May 24 00:51:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
`

sanduan168, 2024-04-23:
[image]
Here is the dufo output.

Kin-Zhang, 2024-04-23:
could you try to clean build, then build again, copy the build information? like following is mine when I run `cmake -B build -D CMAKE_CXX_COMPILER=g++-10 && cmake --build build`:

```bash
-- The CXX compiler identification is GNU 10.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/g++-10 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'liblz4'
--   Found liblz4, version 1.9.2
-- Checking for module 'liblzf'
--   Found liblzf, version 3.6
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- TBB found, enabling parallel execution
-- UFOMAP BMI2 instructions disabled
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kin/dufomap/build
[
```

sanduan168, 2024-04-23:
Sure.
[image]
[image]

sanduan168, 2024-04-26:
Could you tell me the version of TBB in your building environment?

Kin-Zhang, 2024-04-26:
The version is:
```bash
libtbb-dev is already the newest version (2020.1-2).
```

sanduan168, 2024-04-30:
Hi, Kin
Thanks you for your help. I have successively reproduce the experimental results in your paper.
The reason why my results are different from yours is that TBB is not turned on. In my opinion, TBB is a tool to boost computing efficiency. However,  it will not affect dufomap results. So you can try it by disable "find_package(TBB QUIET)"  latter.
BTW, your linux environment is 20.04, but my experiment is ubuntu 18.04LTS.  So I can't download the libtbb-dev with version: 2020.1-2. If guys want to try dufomap,  you can do like this:

1. git clone https://github.com/oneapi-src/oneTBB.git
2. git checkout v2020.1
3. make
4. cp {tbb_path}/build/gnu_10.3_cxx11_64_relwithdebinfo/* {dufomap_path}/ufomap/third_part/tbb/lib
5. modify {dufomap_path}/ufomap/CMakeLists.txt:
[image]

closed 2024-04-30 by Kin-Zhang