KTH-RPL/dufomap #4: dynamic removal result between octomap and dufomap

sanduan168, 2024-05-07:
Hi Qing Wen
I test dufomap and octomap with my dataset.
[image]
The result above is from dufomap with the default config.toml. We can see that some dynamic objects(in red circle) are not trimmed.
The result below is from octomap(https://github.com/Kin-Zhang/octomap.git) with default config.yml. Even though the static objects(in green rectangle) are removed, the dynamic objects are almost deleted.
[image]
In your paper, the SA\DA of dufomap performs better than octomap, which is inconsistent with my experiment. Can you explain why? 
If you want to reproduce, you can download the dataset from: https://drive.google.com/drive/folders/1kpevt1zkFlDNiw-h6ISCD13-W8EdX6By?usp=drive_link

Kin-Zhang, 2024-05-09:
Hi,

the dataset link download is not available. Please give full access: You need access.

hard to tell if I cannot open the data you provided, the conclusion we summarise is based on experiments of existing datasets, when changing datasets/scenarios, the statement from the previous may need to change. That's why all papers will say: based on xxx dataset, we have xxx conclusion. or existing experiment on xxx dataset.

sanduan168, 2024-05-09:
the following link is available:
https://drive.google.com/drive/folders/1kpevt1zkFlDNiw-h6ISCD13-W8EdX6By?usp=sharing

qimao7213, 2024-05-10:
你可以把assert里面的配置文件、resolution调成0.2，然后把ray_passthrough_hits设置为true试试呢。

closed 2024-06-05 by Kin-Zhang