KTH-RPL/dufomap #5: 你好，请问queryNearest函数怎么使用？

Rick-960123, 2024-05-27:


Rick-960123, 2024-05-27:
if constexpr (IsOccupancyMap<Map>) {
			float dis = (*cur - sensor_origin).norm();
			float radius = 5.238 * 1e-4 * dis * dis - 3.23 * 1e-3 * dis + 0.08;
			ufo::AABC aabc(*cur, radius);
			ufo::BoundingVolume bv(aabc);
			for (auto & node_ : map.queryNearest(map.toCode(*cur), bv, ufo::pred::TRUE())) {
				map.updateOccupancyLogit(node_, prob);
			}
			// map.updateOccupancyLogit(node, prob);
		}
我想在更新概率时，也更新周围体素的概率，不知道代码该怎么写，我对c++20不太熟

Kin-Zhang, 2024-06-05:
这个问题建议在 https://github.com/UnknownFreeOccupied/ufomap 里询问，与dufomap主算法无关；

我也没有尝试过ufomap下的概率更新 （Maybe在 https://github.com/UnknownFreeOccupied/ufomap/tree/master/ufomap_ros 这里面的应用里你能找到旧版的使用）