KTH-RPL/dufomap #10: 关于论文中的limizations

whw-create, 2024-07-17:
你好，我在拜读论文时，看到最后的局限性部分说，如果在点云地图边界上的体素，虽然是动态体素，但是由于没有fully observed，也无法被划分为动态体素。我在测试数据时可能遇到了类似的问题，
[image]
测试时动态点云下面一半被过滤掉了，但是路面上方仍然有一串漂浮的点云，不知道有没有什么解决方案呢

whw-create, 2024-07-17:
https://www.alipan.com/t/9EROcChDVr4QZykpOj9u ，你康康链接有效吗，谢谢咯。

Kin-Zhang, 2024-07-17:
ok 有效的，我刚刚快速看了一下整体数据，是limitation的part

具体我可以后续运行一下再返还给你（尽量这周内）一个clean map；我后期做了一个针对这个点的[部分coding](https://github.com/KTH-RPL/SeFlow) 不过还没优化速度

whw-create, 2024-07-17:
啊我只是测试了一点数据，发现了这个问题，不需要你专门给我发一个clean map，我想请教一下的是有没有什么解决办法，我看你论文提到了可以做一下聚类，但是聚类那个代码我实在是看不懂～，归根结底是我看不懂ufomap哈哈，或者请教一下removert、peopleremover和erasor可以应对这种情况吗?

Kin-Zhang, 2024-07-17:
hahah 代码里的聚类不是论文里的聚类那一块的，代码里的聚类是在map里的，具体有出报告了后 我可以再link过来~ 

据我所知，如果精心调参的话erasor 高度参数，ok？具体我可能需要去运行看一下；
因为你的数据已经转了benchmark的形式，所以：https://github.com/KTH-RPL/DynamicMap_Benchmark 这个链接下的所有方法我都做了adaptation，你感兴趣都可以直接运行试试 ^v^

Kin-Zhang, 2024-07-17:
[image]

我看数据播放 这个意思应该是两个lidar？所以dufomap的输入sensor center按你现有的数据走传统处理方式应该有问题 lol...

whw-create, 2024-07-17:
看起来ray casting确实是应该在每一个lidar对应的坐标系下进行的，如果把两个lidar的数据投到同一个坐标系下，做ray casting时扫过的体素就和原始射线的体素不同了，对吗？

Kin-Zhang, 2024-07-17:
对的 你的数据 所有点来自不同的两个中心，每个点来自来个lidar就按哪个中心做，不然会有问题（特别是这两个lidar相距比较远，如果是lidar相距很近是可能就忽略了）

我直接运行了一下我们的离线方案 https://github.com/MKJia/BeautyMap 感觉效果也不错 （没调参... 所以有些破洞的地方 hahah

[image]

whw-create, 2024-07-17:
okok，谢谢啦

closed 2024-07-18 by Kin-Zhang

whw-create, 2024-07-25:
哈喽哈喽，我最近想在一个ros2项目中使用dufomap，但是我一旦引入ros2相关头文件，就会引发一串编译错误：
[image]
有没有碰到过类似的问题呢