KTH-RPL/dufomap #12: ros版本？

maomao0220, 2024-09-03:
您好，如果想改成ros版本，请问如何修改？

Kin-Zhang, 2024-09-03:
不太确定ros版本的需求是什么，如果是paper的任务，不需要ros版本，直接slam跑完结果的bag 传出pcd数据即可，

可以参考这个教程：https://kth-rpl.github.io/DynamicMap_Benchmark/data/creation/#custom-data

maomao0220, 2024-09-03:
您好，就是想改成订阅lidar点云话题和里程计，实现实时的动态点云去除

Kin-Zhang, 2024-09-04:
可以参考这个issue下的tips进行ros的版本修改：

https://github.com/KTH-RPL/DynamicMap_Benchmark/issues/10#issuecomment-2059111679

maomao0220, 2024-10-17:
您好，在改成ROS版本进行建图时，有几个问题：
1、在回调函数中仅使用了ufo::insertPointCloud（进行建图，）发现随着不断的运行处理时间在不断的变大，请问有什么优化方法或者建议吗？
2、请问需不需要使用cluster（）函数，该函数主要是用来干嘛的？

Kin-Zhang, 2024-10-17:
1、原版本无ros的我运行过全sequence数据集，大概4000多frame，并没有这个现象发生
2、如果是复现论文的分数，并不需要，只是用来网页版第一个视频的可视化不同动态障碍物（不同的颜色的）

maomao0220, 2024-10-18:
感谢您的及时回复！
1、通过打印时间耗时，发现耗时主要发生在从ufomap中提取静态点并通过ros话题发布，打印的耗时结果及相应的提取静态点云并发布的代码如下：
[image]
[image]
请问，我这里提取静态点云部分有没有什么问题？有没有解决方法来降低点云提取部分的耗时？能否只提取当前帧插入到地图中的点云而不是遍历整张ufomap？

2、在建图过程中，发现没有去除对面车辆形成的拖影，其中车辆为大型车辆（大概5、6米高），请问需要调节哪些参数？
[image]
[image]

非常期待您的回复

Kin-Zhang, 2024-10-19:
1、如果是建图任务，没必要每一帧都要读取所有地图；建议保持这个repo一样的类型，在最后get map就行
2、需要拿到你的数据我才能仔细给出意见

maomao0220, 2024-10-28:
您好，因为我在测试过程中，还有其他模块需要地图的实时输出结果，所以想着以10Hz频率实时输出点云地图，请问有什么建议吗？代码中我只查询当前帧点云而不是累积的点云，但是建图效果不太好，类似拼接点云效果，出现了很多毛边