KTH-RPL/dufomap #16: Static Points remove too much

zjulabwjt, 2024-12-13:
I use custom driving dataset remove dynamic points, use assert folder config, but I found points around ego car (specially on the ground) is removed. Can you give me some advice to solved this problem?Thank you!

Kin-Zhang, 2024-12-13:
It's hard to tell without taking a look at your data. If it's possible could you provide more visualization or data directly?

zjulabwjt, 2024-12-16:
> It's hard to tell without taking a look at your data. If it's possible could you provide more visualization or data directly?

@Kin-Zhang  thanks for your quickly reply!
[image]
[image]
First image is pcd after remove dynamic points, and second image is origin pcd! The ground points around ego car is removed too much. I used muti lidars pointcloud and had set different viewpoint in trajectory.Can you give me some advice?Thank you!

Kin-Zhang, 2024-12-16:
If that possible to share the data with me and let me know what's your config on it so I can reproduce your results?

zjulabwjt, 2024-12-18:
> If that possible to share the data with me and let me know what's your config on it so I can reproduce your results?

Thank you for your quick reply! However, it seems inconvenient for me to provide data. May I ask you a few related questions?

1.In my experiments in an underground parking scene, I noticed that a large number of ground points are mistakenly removed on ramps and turns. Is this caused by incorrect parameter settings?

2.Can DUFOMAP handle point cloud results from a single frame?

3.Regarding the issue of ground points being mistakenly removed in the my data, I am currently segmenting the ground and then performing dynamic object removal on non-ground points to avoid the mistake. However, it seems that points near the ground cannot be removed. Is this phenomenon reasonable?

MichealRW, 2024-12-18:
I hope this message finds you well. I am currently working on reproducing the `dufomap` algorithm that you have provided on GitHub, and I have encountered some issues that I would greatly appreciate your assistance with.

Specifically, when I use the full set of frames from the KITTI dataset as input, the algorithm's output does not meet the expected results. Despite using the same methods for generating the ground truth (GT), the point cloud generation method from the evaluated algorithm, and the evaluation methodology as described in the `DynamicMap_Benchmark`, the final results are not satisfactory, with particularly low DA values. However, when I test the algorithm with only a subset of the frames, the results are more in line with those presented in the paper.
[image]

I would greatly appreciate any insights you might have regarding potential issues or suggestions for debugging or improving the implementation.

Thank you for your time and assistance. I look forward to hearing from you.

Best regards,

Kin-Zhang, 2024-12-21:
> 1.In my experiments in an underground parking scene, I noticed that a large number of ground points are mistakenly removed on ramps and turns. Is this caused by incorrect parameter settings?

Sorry to say this, but it's really hard to tell if I don't have your data. Especially I don't want if you process data correctly.... For DUFOMap I tested on multiple floor (7 floor) in HKUST Academic hall, and the result was still good enough. So I don't think it's the algorithm problem but may data process.

> 2.Can DUFOMAP handle point cloud results from a single frame?

Yes, we have kind of single frame result in Python API even. Check [this work](https://github.com/KTH-RPL/SeFlow) and [Python script](https://github.com/KTH-RPL/SeFlow/blob/main/process.py).

> 3.Regarding the issue of ground points being mistakenly removed in the my data, I am currently segmenting the ground and then performing dynamic object removal on non-ground points to avoid the mistake. However, it seems that points near the ground cannot be removed. Is this phenomenon reasonable?

I would strongly recommend octomap w fg: https://github.com/Kin-Zhang/octomap. It's our first try on ray-casting with noise filter and ground segmentation involved. However, I didn't engineer optimized runtime. So the runtime is quite slow compared to DUFOMap. And you can reference the link I provided and see how we do non-ground points but still use ground points to do ray-casting.