KTH-RPL/dufomap #17: 关于单帧分割结果

Amnesiacers, 2024-12-17:
hi，非常棒的工作！

我想得到每帧pcd点云每个点的分割结果，读过代码决定在ic分离misses之后，将余下的点当做hits来处理，最后将misses和hits统一转为xyz的格式进行保存，不过后来发现类型转换实在是有点让人眩晕orz

想请问下这种处理方法是正确的吗，以及有什么更好的建议吗

Amnesiacers, 2024-12-17:
尝试了一种更简单的办法，在处理完当前帧后，把当前cloud扔回map进行seenFree的判别，平均运行时间多了40ms左右，但代码改动量较少

Kin-Zhang, 2024-12-21:
Hi 谢谢认可

> 读过代码决定在ic分离misses之后，将余下的点当做hits来处理，最后将misses和hits统一转为xyz的格式进行保存，不过后来发现类型转换实在是有点让人眩晕orz

完全理解，更好的建议就是你第二条提出的（至少我在我的新工作中是这样识别的：https://github.com/KTH-RPL/SeFlow） 这里面我对单帧的分割结果进行保存，作为prior 到自监督学习用

> 平均运行时间多了40ms左右，但代码改动量较少

运行时间我记得纯C++的话 好像是会多一点，但是我不太记得清是否有到40ms；代码的话 如果你不想改，可以直接使用Python API我上面提到的repo 这个script：https://github.com/KTH-RPL/SeFlow/blob/main/process.py

Amnesiacers, 2025-01-21:
这是一个迟到的回复
非常感谢张博耐心的解答，感谢整个Benchmark，解决了非常多脏活累活blabla
祝你科研顺利，生活开心！ ^_^

closed 2025-02-19 by Kin-Zhang