KTH-RPL/dufomap #20: 使用DUFOMap处理Livox mid360数据时遇到的问题

LC-Robotics, 2025-02-18:
在使用DUFOMap处理我使用Livox mid360采集的室内数据(多层楼梯环境)时，输出的点云图似乎没有滤除任何动态点，与所有点云累积起来基本一样。我使用脚本对比我标注的ground truth以分析保留率和拒绝率，结果是保留率100%，而拒绝率0%，这似乎也说明了输出的点云图没有滤除任何动态点。我的数据可以通过下面的链接获得：

通过网盘分享的文件：pcd.zip
链接: https://pan.baidu.com/s/1TUaF4trT_vYbBJrOjzJzdw 提取码: b6mi

奇怪的是DUFOMap在我使用robosense rs-32采集的室内数据集上运行的很好，保留率和拒绝率都大于90%。此外我还在SemanticKITTI和HeLiMOS的四种传感器上运行了DUFOMap，却都没有出现过这种问题。我已经根据Fast-lio2得到的位姿将点云变换到世界坐标系下，并在sensor origin添加了view point，似乎所有过程都没有问题。我尝试过调整推荐调整的那三个参数，但没有得到有意义的结果。

我已经在数个动态点云滤除的算法(包括beautymap)中测试过使用Livox mid360采集的室内数据，保留率和拒绝率都表现正常(beautymap的数据较低，似乎是不擅长处理这种多层结构)

感谢您的帮助！

Kin-Zhang, 2025-02-19:
Hi, 谢谢提供数据，我看了一下 问题出现在数据的 0,0,0 sensor 发出point的地方 有一个pose point (这个问题我在eth那边数据也看到了) 主要是他们为了可视化pose做的；我做了一次近距离过滤就好了，以下是效果图：

[image]

具体测试的完整代码（能运行出上述截图的代码）：环境只要有个python>=3.8的坏境，然后 `pip install dufomap` 即可配置完成 直接运行下面代码

```python
"""
# Created: 2024-11-20 13:11
# Copyright (C) 2024-now, RPL, KTH Royal Institute of Technology
# Author: Qingwen Zhang  (https://kin-zhang.github.io/)
#
# This file is part of DUFOMap (https://github.com/KTH-RPL/dufomap) and 
# DynamicMap Benchmark (https://github.com/KTH-RPL/DynamicMap_Benchmark) projects.
# If you find this repo helpful, please cite the respective publication as 
# listed on the above website.

# Description: Output Cleaned Map through Python API.
"""
from pathlib import Path
import os, fire, time
import numpy as np
from tqdm import tqdm

from dufomap import dufomap
from dufomap.utils import pcdpy3
def inv_pose_matrix(pose):
    inv_pose = np.eye(4)
    inv_pose[:3, :3] = pose[:3, :3].T
    inv_pose[:3, 3] = -pose[:3, :3].T.dot(pose[:3, 3])
    return inv_pose

MIN_AXIS_RANGE = 0.5 # HARD CODED: remove ego vehicle points
MAX_AXIS_RANGE = 50 # HARD CODED: remove far away points

class DynamicMapData:
    def __init__(self, directory):
        self.scene_id = directory.split("/")[-1]
        self.directory = Path(directory) / "pcd"
        self.pcd_files = [os.path.join(self.directory, f) for f in sorted(os.listdir(self.directory)) if f.endswith('.pcd')]

    def __len__(self):
        return len(self.pcd_files)
    
    def __getitem__(self, index_):
        res_dict = {
            'scene_id': self.scene_id,
            'timestamp': self.pcd_files[index_].split("/")[-1].split(".")[0],
        }
        pcd_ = pcdpy3.PointCloud.from_path(self.pcd_files[index_])
        pc0 = pcd_.np_data[:,:3]
        res_dict['pc'] = pc0.astype(np.float32)
        res_dict['pose'] = list(pcd_.viewpoint)
        return res_dict

def main_vis(
    data_dir: str = "/home/kin/data/dufo_issue20",
):
    dataset = DynamicMapData(data_dir)

    # STEP 0: initialize 
    mydufo = dufomap(0.1, 0.1, 1, num_threads=12) # resolution, d_s, d_p same with paper.
    cloud_acc = np.zeros((0, 3), dtype=np.float32)
    for data_id in (pbar := tqdm(range(0, len(dataset)),ncols=100)):
        data = dataset[data_id]
        now_scene_id = data['scene_id']
        pbar.set_description(f"id: {data_id}, scene_id: {now_scene_id}, timestamp: {data['timestamp']}")
        norm_pc0 = np.linalg.norm(data['pc'][:, :3] - data['pose'][:3], axis=1)
        range_mask = (
                (norm_pc0>MIN_AXIS_RANGE) & 
                (norm_pc0<MAX_AXIS_RANGE)
        )
        # STEP 1: integrate point cloud into dufomap
        mydufo.run(data['pc'][range_mask], data['pose'], cloud_transform = False)
        # STEP 1: integrate point cloud into dufomap
        cloud_acc = np.concatenate((cloud_acc, data['pc']), axis=0)
    
    # STEP 2: propagate
    mydufo.oncePropagateCluster(if_propagate=True, if_cluster=False)
    # STEP 3: Map results
    mydufo.outputMap(cloud_acc, voxel_map=False)
    # NOTE(Qingwen): You can also save voxeled map directly based on the resolution we set before:
    # mydufo.outputMap(cloud_acc, voxel_map=True)
    
    mydufo.printDetailTiming()

if __name__ == "__main__":
    start_time = time.time()
    fire.Fire(main_vis)
    print(f"Time used: {time.time() - start_time:.2f} s")
```

Kin-Zhang, 2025-02-19:
如果还有后续问题 可以留言哈，我最近赶ddl中；有空就会回复 😊；祝好~

LC-Robotics, 2025-02-21:
赶ddl还能这么快回复，真的非常感谢😊。确实就是pose point的问题，只要滤除近距离点就好了。

closed 2025-02-21 by LC-Robotics