KTH-RPL/dufomap #23: How to Run DUFOMap in Online Mode?

windzu, 2025-03-05:
Hello,

First of all, thank you for your open-source work! I followed your tutorial and tested with custom data, achieving good results.

However, my use case is quite different. I noticed in your documentation that DUFOMap supports online usage, which is exactly what I need.

My understanding of online mode is that the system can build and output the map in real-time based on incoming point cloud data, whereas offline mode means reading all point cloud data at once and then generating a final map.

Based on the provided code, it seems like only the offline implementation is available, and I couldn't find an online version.

So my question is:
How can I perform online testing using the existing code?

Additionally, in online mode, is the map able to update dynamically for objects that are initially stationary but start moving later? For example, a car waiting at a traffic light that eventually moves?

My goal with this method is to achieve real-time representation of static obstacles using only LiDAR. Then, by integrating an obstacle detection model, I could identify dynamic obstacles, potentially creating a more robust LiDAR perception system.

What are your thoughts on this approach?

Looking forward to your response!

Best regards

Kin-Zhang, 2025-03-30:
> How can I perform online testing using the existing code?

In the paper online testing is send data step by step and detection dynamic with current map info. I think you can do the scan query inside for data.

> Additionally, in online mode, is the map able to update dynamically for objects that are initially stationary but start moving later? For example, a car waiting at a traffic light that eventually moves?

I would say it's case by case, but yes. If it eventually moves, it will be detected as dynamic when it moves but not when it is static if you run online. But if it's offline, it can be removed by dufomap.

Kin-Zhang, 2025-03-30:
For the online testing, I can try later if I have time to have ROS1/ROS2 version so people can real run in robotics system.... But the task is more like detect and **remove** dynamic, as you can also see from paper it's not designed for perception task I would say.

closed 2025-09-26 by github-actions[bot]