KTH-RPL/dufomap #30: Seeking guidance on parameter tuning for custom dataset - suboptimal dynamic removal results

WinRobotics, 2025-08-17:
Hi Dr. Kin-Zhang,

Thank you very much for open-sourcing your code. I am currently working on creating a ROS2 bag conversion tool to extract data into the PCD format required by DUFOMap, and I plan to share it once I’ve completed testing.

I’ve tried running DUFOMap with data generated by my script, but the results is very bad (i felt like i did something wrong somewhere but i tried many days to debug). I’d like to seek your advice on whether this might be due to issues with my custom dataset, or if I may be misunderstanding certain aspects of the DUFOMap algorithm.

In particular, I’ve recorded some ROS2 bag sessions with a colleague walking around, and my goal is to remove him as a dynamic obstacle from the map. I would greatly appreciate any guidance you could provide on this.

The PCD file generated from my pcd_ros2bag python script is located inside here. Using the standard perimeters from your paper.

https://drive.google.com/drive/folders/1oVPn97z2RN7J0cM6niF4M24zvK5AugZ4?usp=sharing

Best regards,
Sherwin

Kin-Zhang, 2025-08-17:
I redo it for you as follows:

1. Convert ROS2 bag to ROS1 bag, through this: https://gitlab.com/ternaris/rosbags
2. Record your point cloud topic as a new bag (since I don't know why your odom and tree topic timestamp is wrong.
3. Use my simple_ndt_slam to get the odom.

[image]

4. Extract bag to pcd folder file: https://github.com/Kin-Zhang/simple_ndt_slam/tree/master/tools#extract_pcdfbag_cpp. And check the raw map is correct:

[image]

6. Process with DUFOMap (remove your ego point first, as shown in the following play data, you have a few ego points near the agent which block the ray casting. Check the similar issue here: https://github.com/KTH-RPL/dufomap/issues/20)

[image]

---

Here are the DUFOMap result (default param in this repo) and raw map on the right.:

[image]

WinRobotics, 2025-08-17:
Thanks so much for the input and help !

WinRobotics, 2025-08-22:
Hi Dr Kin-Zhang,

I’d like to confirm I followed your instructions: I re-recorded the ROS 2 bag (rosbag2) into ROS 1 with only the LiDAR cloud topic, then ran roslaunch lidar_localizer ndt_mapping_docker.launch to generate the /leg_odom topic, and finally recorded a new bag that includes the odometry topic.

These are the topics that my new bag has:
/auto_odom
/clicked_point
/clock
/current_pose
/imu
/initialpose
/leg_odom
/lidar_points_filtered
/move_base_simple/goal
/ndt_map
/offset_leg_odom
/repub_points
/robot_path
/rosout
/rosout_agg
/tf
/tf_static

Next, I’ll convert the bag to PCD files using ./bag2pcd <rosbag_path> <save_pcd_folder> <odom_topic> <cloud_topic> [1 to also save the raw map]. After I have both the global/raw PCD and the per-frame PCDs, I’ll run DUFOMap (Step 6) to remove any residual ego points.

Would that be correct?

Kin-Zhang, 2025-08-22:
> roslaunch lidar_localizer ndt_mapping_docker.launch to generate the /leg_odom topic

the correct one is auto_odom, please check [the screenshot in my previous comment](https://github.com/KTH-RPL/dufomap/issues/30#issuecomment-3194634145) where I show the full command about. Follow the comment steps it should works as I showed.

closed 2025-09-21 by WinRobotics

WinRobotics, 2025-11-15:
Hi Kin Zhang,

I realized my mistake, and after reviewing my data, I now understand that I should have used TF for the transformation step. I’ve been referring to your implementation here:

https://github.com/Kin-Zhang/simple_ndt_slam/blob/master/tools/src/extract_pcdFbag_tf.cpp

I’m currently working on a ROS 2 version so that others can more easily test this on ROS 2 as well.

Apologies for the disturbance, and thank you for your work.